fewieden/MMM-voice #4: Mode selection

fragiskos89, 2017-02-22:
I'm sorry I post this as an issue but I don't see another way to contact you. 
I am trying to find a way to bypass the mode selection phase and just straight to VOICE mode.
I would be really thankful if you could point me in the right direction.

fewieden, 2017-02-23:
this is not possible without modifying the code, I added this to avoid some false positives, but if you want you can change this line https://github.com/fewieden/MMM-voice/blob/f9eb4cd4bde5c64ef894e8625250c4092aa0d4a7/node_helper.js#L18 to ```mode: "VOICE",```

fragiskos89, 2017-02-23:
Thanks for the reply. Will give it a try tomorrow and answer here if the result was what I wanted. :)

fragiskos89, 2017-02-24:
Ok so I did what you said (again thanks for helping) but it doesn't seem to change anything. The module still says "Ready, choose a mode".
Sorry if I'm being a bother but as you have probably guessed I'm new to this.

fewieden, 2017-02-24:
this is just the ui it still works without forcing the user to choose a mode, but if you want that change the line https://github.com/fewieden/MMM-voice/blob/f9eb4cd4bde5c64ef894e8625250c4092aa0d4a7/MMM-voice.js#L35

fragiskos89, 2017-02-24:
Ohh ok... I couldn't try it because my microphone didn't seem to want to cooperate. Thanks again for the help. I'll close this issue now.

closed 2017-02-24 by fragiskos89