fewieden/MMM-voice #8: ERROR on loading Module

moFries, 2017-04-09:
Hi,

today I tried to run your module, but all I get is an error messsage on my mirror:

Message on mirror (gif):
[image]

Message on terminal:
[image]

I took a look in error.log, it says "**127**".

First I thought, my microphone is not found, but: 

```pi@rp_mm:~/MagicMirror $ cat /proc/asound/cards
 0 [ALSA           ]: bcm2835 - bcm2835 ALSA
                      bcm2835 ALSA
 1 [Device         ]: USB-Audio - USB PnP Sound Device
                      C-Media Electronics Inc. USB PnP Sound Device at usb-3f980000.usb-1.4, full
```

Config is:
>                 {
>                         module: 'MMM-voice',
>                         position: 'bottom_bar',
>                         config: {
>                                 microphone: 1,
>                                 // ...
>                         }
>                 },

So my question is: Do you have an idea, why it' not working. I wasn't able to find out, what error "127" could mean.

Thx in advance

moFries

fewieden, 2017-04-09:
those errors come from pocketsphinx itself.

Can you check the debug log with enabling `debug: true` in the config?

moFries, 2017-04-09:
Das gesamte Ergebnis steht weiter unten.

Das wichtigste war wohl: `Error opening audio device plughw:1 for capture: Device or resource busy`

Ich konnte mir nicht vorstellen, wieso das sein konnte. Naja, ich habe dann einmal in der GUI das MIkrofin als "Default" ausgewählt und voila, beim nächsten Start ging es. Ich werde es jetzt mal beobachten, wie es sich weiterentwickelt die Tage. 

Vielen Dank schon mal.

> INFO: pocketsphinx.c(152): Parsed model-specific feature parameters from /usr/local/share/pocketsphinx/model/en-us/en-us/feat.params
> Current configuration:
> [NAME]			[DEFLT]		[VALUE]
> -agc			none		none
> -agcthresh		2.0		2.000000e+00
> -allphone				
> -allphone_ci		yes		yes
> -alpha			0.97		9.700000e-01
> -ascale			20.0		2.000000e+01
> -aw			1		1
> -backtrace		no		no
> -beam			1e-48		1.000000e-48
> -bestpath		yes		yes
> -bestpathlw		9.5		9.500000e+00
> -ceplen			13		13
> -cmn			live		batch
> -cmninit		40,3,-1		41.00,-5.29,-0.12,5.09,2.48,-4.07,-1.37,-1.78,-5.08,-2.05,-6.45,-1.42,1.17
> -compallsen		no		no
> -debug					0
> -dict					modules/MMM-voice/MMM-voice.dic
> -dictcase		no		no
> -dither			no		no
> -doublebw		no		no
> -ds			1		1
> -fdict					
> -feat			1s_c_d_dd	1s_c_d_dd
> -featparams				
> -fillprob		1e-8		1.000000e-08
> -frate			100		100
> -fsg					
> -fsgusealtpron		yes		yes
> -fsgusefiller		yes		yes
> -fwdflat		yes		yes
> -fwdflatbeam		1e-64		1.000000e-64
> -fwdflatefwid		4		4
> -fwdflatlw		8.5		8.500000e+00
> -fwdflatsfwin		25		25
> -fwdflatwbeam		7e-29		7.000000e-29
> -fwdtree		yes		yes
> -hmm					/usr/local/share/pocketsphinx/model/en-us/en-us
> -input_endian		little		little
> -jsgf					
> -keyphrase				
> -kws					
> -kws_delay		10		10
> -kws_plp		1e-1		1.000000e-01
> -kws_threshold		1e-30		1.000000e-30
> -latsize		5000		5000
> -lda					
> -ldadim			0		0
> -lifter			0		22
> -lm					modules/MMM-voice/MMM-voice.lm
> -lmctl					
> -lmname					
> -logbase		1.0001		1.000100e+00
> -logfn					
> -logspec		no		no
> -lowerf			133.33334	1.300000e+02
> -lpbeam			1e-40		1.000000e-40
> -lponlybeam		7e-29		7.000000e-29
> -lw			6.5		6.500000e+00
> -maxhmmpf		30000		30000
[... 86 lines cut ...]
> INFO: dict.c(213): Dictionary size 39, allocated 0 KiB for strings, 0 KiB for phones
> INFO: dict.c(361): 5 words read
> INFO: dict2pid.c(396): Building PID tables for dictionary
> INFO: dict2pid.c(406): Allocating 42^3 * 2 bytes (144 KiB) for word-initial triphones
> INFO: dict2pid.c(132): Allocated 21336 bytes (20 KiB) for word-final triphones
> INFO: dict2pid.c(196): Allocated 21336 bytes (20 KiB) for single-phone word triphones
> INFO: ngram_model_trie.c(354): Trying to read LM in trie binary format
> INFO: ngram_model_trie.c(365): Header doesn't match
> INFO: ngram_model_trie.c(177): Trying to read LM in arpa format
> INFO: ngram_model_trie.c(193): LM of order 3
> INFO: ngram_model_trie.c(195): #1-grams: 29
> INFO: ngram_model_trie.c(195): #2-grams: 54
> INFO: ngram_model_trie.c(195): #3-grams: 27
> INFO: lm_trie.c(474): Training quantizer
> INFO: lm_trie.c(482): Building LM trie
> INFO: ngram_search_fwdtree.c(74): Initializing search tree
> INFO: ngram_search_fwdtree.c(101): 31 unique initial diphones
> INFO: ngram_search_fwdtree.c(186): Creating search channels
> INFO: ngram_search_fwdtree.c(323): Max nonroot chan increased to 168
> INFO: ngram_search_fwdtree.c(333): Created 31 root, 40 non-root channels, 5 single-phone words
> INFO: ngram_search_fwdflat.c(157): fwdflat: min_ef_width = 4, max_sf_win = 25
> INFO: continuous.c(307): pocketsphinx_continuous COMPILED ON: Apr  9 2017, AT: 11:29:15
>
> Error opening audio device plughw:1 for capture: Device or resource busy
> FATAL: "continuous.c", line 245: Failed to open audio device

trussellaaa, 2017-04-12:
I'm having the same problem - exactly as above.

rcuart, 2017-04-20:
I got the same ERROR in the screen and when i try to debug with "debug;true" a "undefined" message appear behind the "ERROR" one.

fewieden, 2017-04-20:
@trussellaaa @rcuart the translated solution of @moFries was to set the microphone in the pi gui to default

@rcuart if debug is enabled the undefined message is expected before any voice is recognized, but as you have an error this will not change. can you share your debug.log file

rcuart, 2017-04-23:
@fewieden thanks for your fast answer!
Here's my 'debug.log' file:
`pocketsphinx_continuous: error while loading shared libraries: libpocketsphinx.so.3: cannot open shared object file: No such file or directory`

fewieden, 2017-04-23:
It cannot find the installation of pocketsphinx.
Can you check your bashrc file if there is the following entry at the bottom?
```
LD_LIBRARY_PATH=/usr/local/lib
PKG_CONFIG_PATH=/usr/local/lib/pkgconfig
```
Did you run the installer with sudo rights?

rcuart, 2017-04-23:
mmm i'll try to reinstall with sudo rights.
I'm giving feedback this evening!

Thanks.

rcuart, 2017-04-23:
`[STEP 3/6] Installing pocketsphinx | Done
[STEP 4/6] Exporting paths
[STEP 4/6] Exporting paths |  Done
[STEP 5/6] Installing npm dependencies
dependencies.sh: línea 106: cd: /root/MagicMirror/modules/MMM-voice: No existe el fichero o el directorio
npm WARN enoent ENOENT: no such file or directory, open '/root/pocketsphinx/package.json'
npm WARN pocketsphinx No description
npm WARN pocketsphinx No repository field.
npm WARN pocketsphinx No README data
npm WARN pocketsphinx No license field.
[STEP 5/6] Installing npm dependencies | Done
[STEP 6/6] Manipulating dependencies
dependencies.sh: línea 118: cd: /root/MagicMirror/modules/MMM-voice/node_modules/pocketsphinx-continuous: No existe el fichero o el directorio
sed: no se puede leer index.js: No existe el fichero o el directorio
[STEP 6/6] Manipulating dependencies | Failed
`

fewieden, 2017-04-23:
I just asked if you did it. Reason is that you shouldn't do it :D

rcuart, 2017-04-23:
Right! The problem is that i realized i was wrong too late -_-.
Now making a clean install of the OS, mirror and then the module.

rcuart, 2017-04-25:
Now i got the error '127' too.
Debug:

`pocketsphinx_continuous: error while loading shared libraries: libpocketsphinx.so.3: cannot open shared object file: No such file or directory
`

rcuart, 2017-04-25:
SOLVED:
`LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/usr/local/lib`
`sudo ldconfig`

More info: [http://stackoverflow.com/questions/4743233/is-usr-local-lib-searched-for-shared-libraries](url)

fewieden, 2017-04-25:
the installer should have placed that in the bashrc in the first place

closed 2017-04-25 by fewieden

fewieden, 2018-04-26:
@marktillo you might want to get in touch with someone here https://forum.magicmirror.builders/ I know that several people on the forum are using ps3 eye camera

shazglas02, 2018-06-04:
How do I uninstall MMM-voice as it just shows an ERROR even after running the commands you mentioned above!?

fewieden, 2018-06-04:
@shazglas02 https://forum.magicmirror.builders/topic/778/mmm-voice/101

Jacobphilli, 2018-08-30:
@rcuart @fewieden 

An above comment says this: 

“SOLVED:
LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/usr/local/lib
sudo ldconfig

More info: http://stackoverflow.com/questions/4743233/is-usr-local-lib-searched-for-shared-libraries”

I looked at the link included but it’s still not clear what I should do. I’m also receiving this error message and would like to fix it. 

Could you help me out? Thanks!

fewieden, 2018-08-30:
Execute those two commands in your terminal

nabbi95, 2019-02-10:
still not working