fewieden/MMM-voice #10: LANG: Spanish

rcuart, 2017-04-25:
Ey!
Now that i got it working, i want to use commands in spanish.
How could i do that? Do i have to re-write de dictionary and the rest of files?

Thanks!

fewieden, 2017-04-27:
This would be a rather complex process.

First of all you have to install the spanish models for pocketsphinx.
Then you have to generate the dictionairy and language model.
This didn't work properly for most of the words in english see #3, thats why I changed it to use an online service.
Last thing the module is not designed to be used in a different language as the commands are coded in english.

rcuart, 2017-04-27:
Okey, thanks!

fewieden, 2017-04-27:
If you want to play around these 3 links can be helpful:
* http://cmusphinx.sourceforge.net/wiki/tutorialdict
* http://cmusphinx.sourceforge.net/wiki/tutoriallm
* https://sourceforge.net/projects/cmusphinx/files/Acoustic%20and%20Language%20Models/

closed 2017-04-28 by fewieden