fewieden/MMM-voice #12: Problem when start MM

ezar, 2017-05-28:
Platform (Hardware/OS):
Reaspberry Pi 2 / Jessie Lite

Node version:
v6.10.3

MagicMirror version:
2.1.1.

Module version:
GIT CLONE

Description of the issue:
1|mm       | TypeError: this.sendSocketNotification is not a function
1|mm       |     at fs.appendFile (/home/pi/MagicMirror/modules/MMM-voice/node_helper.js:311:22)
1|mm       |     at /home/pi/MagicMirror/modules/MMM-voice/node_modules/graceful-fs/graceful-fs.js:43:10
1|mm       |     at FSReqWrap.oncomplete (fs.js:112:15)

fewieden, 2017-05-29:
This module doesn't use `graceful-fs`, did you install that manually?

ezar, 2017-05-29:
No. I install MMM-voice using dependencies script.

ezar, 2017-05-30:
Nothing? Please help me!
Regards,

ezar, 2017-06-03:
Nothing?

fewieden, 2017-06-07:
The new release v1.0.2 should fix that problem #14

ezar, 2017-06-13:
It works!

closed 2017-06-13 by fewieden