fewieden/MMM-voice #13: MMM error when enable debug

ezar, 2017-06-03:
0|mm       | TypeError: Cannot read property 'debug' of undefined
0|mm       |     at PocketSphinxContinuous.handleData (/home/pi/MagicMirror/modules/MMM-voice/node_helper.js:247:28)
0|mm       |     at emitOne (events.js:96:13)
0|mm       |     at PocketSphinxContinuous.emit (events.js:191:7)
0|mm       |     at Socket.<anonymous> (/home/pi/MagicMirror/modules/MMM-voice/node_modules/pocketsphinx-continuous/index.js:39:11)
0|mm       |     at emitOne (events.js:96:13)
0|mm       |     at Socket.emit (events.js:191:7)
0|mm       |     at readableAddChunk (_stream_readable.js:178:18)
0|mm       |     at Socket.Readable.push (_stream_readable.js:136:10)
0|mm       |     at Pipe.onread (net.js:560:20)

fewieden, 2017-06-07:
The new release v1.0.2 should fix that problem #14

ezar, 2017-06-13:
It works!

closed 2017-06-13 by fewieden