fewieden/MMM-voice #17: "Initialising"

TheRealLinzer, 2017-11-21:
Hey,
first of all many thanks for the awesome module!
I installed everything according to the readme but when I start Magic Mirror it’s displaying “Initialising…” and the mic icon has a slash through it and is blinking infinitely.
I'm using a standard usb-mic and I've already checked for the correct ID.
Using the latest MM-version and updated.

Maybe someone can help? Many thanks in advance :)

fewieden, 2017-12-06:
can you enable `debug: true` in config and then check your logs for more information

ccuster1995, 2018-12-12:
Having the same problem

cowboysdude, 2018-12-16:
Did you enable debug: true in config and then check your logs for more information?  Saying it doesn't work doesn't make it easy to help you..... ;)  So in your config.js under MMM-voice put in debug: true and see what the console log and log tells you then put it here :)

nabbi95, 2019-02-10:
> Did you enable debug: true in config and then check your logs for more information? Saying it doesn't work doesn't make it easy to help you..... ;) So in your config.js under MMM-voice put in debug: true and see what the console log and log tells you then put it here :)

ia m having same problem and i turned debug into true but nothing changes it still shows initialising.
can you tell me where can i get this log file.

nabbi95, 2019-02-10:
even after changing debug mode to true it shows intitialising with written undefined under initialising

fewieden, 2019-02-27:
@nabbi95 the debug flag creates a log file in the module directory

ArroganTz, 2019-08-15:
Im sorry to reopen this issue but i cant find a fix in here and neither do any logs appear/help after i turned on debug: true. Im getting the exact same problem. Initializing.. and under that it says "undefined".

fewieden, 2019-09-11:
@ArroganTz with debug set to true there will be a log file generated in the module directorz. What is the content of this file?

dthomas888, 2019-09-12:
I'm having the same problem: 
error logged this.

pocketsphinx_continuous: error while loading shared libraries: libpocketsphinx.so.3: cannot open shared object file: No such file or directory

dthomas888, 2019-09-12:
I figured it out.
run "sudo ldconfig" after installation to fix.
LD_LIBRARY_PATH needs to be configured.

closed 2021-01-15 by fewieden