fewieden/MMM-voice #18: Adding other voice controls directly into the module

weeee99, 2017-12-22:
Platform (Hardware/OS): RPi3 - Raspian Stretch

Node version: 8.9.3

MagicMirror version: 2

Module version: Latest

Description of the issue: 

as the title said i want to add other voice commands into the module itself (not integrating other modules with voice command) is their any way the codes can be altered and what file to edit exactly, im still new to this and i want to know if its possible thanks :)

fewieden, 2017-12-23:
Actually there is no reason to edit this module for that, just create a new module with the only purpose to handle your commands. then you will be able to update this module without conflicts. out of curiosity, what commands would you like to add? if there reasonable and could be interesting for everyone i might add them anyways.

weeee99, 2017-12-24:
thank for the reply, i was wondering if i can control the GPIOs with this
module alone without  making another module because im still new into
programming raspberry Pis and i want also to save the trouble on making
another one thats why i want to know if the module can be altered to do
commands that can trigger GPIO pins on the Raspi board :)

fewieden, 2017-12-24:
You can but thats the same or even more effort todo

closed 2018-02-02 by fewieden