fewieden/MMM-voice #20: Error when starting

cowboysdude, 2018-01-21:
Platform (Hardware/OS):
Unbuntu 16

Node version:
8.9.4
MagicMirror version:
2.2.2
Module version:
Lastest Master
Description of the issue:
when starting it intializes then goes right to ERROR
When run in debug: true it says undefined.
I ran the command line and my mic is 0.....

Not sure what's undefined but I've tried the mic at 0 and 1......still get ERROR

cowboysdude, 2018-01-22:
Fixed it.  Thanks

fewieden, 2018-01-22:
Could you pls report for other users, what caused the issue und how you fixed it?

cowboysdude, 2018-01-22:
Don’t run them in the MagicMirror directory. Just open a new terminal window and run them
LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/usr/local/lib
and
sudo ldconfig
the 2nd will need your password
and this will verify your microphone number for your config
cat /proc/asound/cards]]

;)

KriyenKP, 2018-01-25:
Still having this issue. Says "Error" on Mirror. But I set the write mic on config. Using a USB PnP Sound Device and it works fine with AlexaPi. Think there's a clash maybe??

fewieden, 2018-01-29:
@KriyenKP i have no experience with alexapi

But you could check the error log for more details

mikeb8319, 2018-04-25:
Not sure how I missed that. Thanks, homie.

closed 2018-04-25 by fewieden

adbeem, 2018-09-24:
hi fewieden, i tried the two commands:
LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/usr/local/lib
sudo ldconfig
but nothing showed up. i check the file directory and i cant seem to find any ldconfig file (also in hidden mode). it is missing somewhere?

adbeem, 2018-09-24:
It starts from initializing then ready choose a mode and immediately goes to ERROR, i checked the error.log it shows 1 1 1 1 1 1 ...