fewieden/MMM-voice #21: How can I best add support for my module?

E3V3A, 2018-03-10:
How can I add support for https://github.com/E3V3A/MMM-FlightsAbove/ ?
I read some of you dev instructions, but I found myself more confused after reading, than before.

fewieden, 2018-03-11:
If you're more detailed about which steps you didn't understand, then I might can explain it to you, as for me the instructions are fine, but someone might not understand it the way i meant.

We also could implement that functionality next week into your module

E3V3A, 2018-03-12:
I just added a notification in my MMM-*.js to support Mykles Hello-Lucy. 

> ...but someone might not understand it the way i meant.

Yeah, I'm definitely a bit confused here. Actually, let me try to install your module first, that will surely help!

closed 2018-04-02 by fewieden