fewieden/MMM-voice #24: Recognize more then one Language.

MattiaFailla, 2018-07-17:
Platform (Hardware/OS): Linux/Raspberry Pi

Node version:

MagicMirror version:

Module version:

Description of the issue: Please, add the possibility to recognize more then one Language, or an HowTo.

fewieden, 2018-07-17:
I don't have the time to support multiple languages, but I guess here would be a good starting point for you https://cmusphinx.github.io/wiki/faq/#q-which-languages-are-supported

MattiaFailla, 2018-07-18:
>I don't have the time to support multiple languages, but I guess here would be a good starting point for you https://cmusphinx.github.io/wiki/faq/#q-which-languages-are-supported

@fewieden

It would be interesting to add the possibility to select the prefer language in the config.js file. I think I can work on it.

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>