fewieden/MMM-voice #25: Help with installation

314noob, 2018-07-26:
Platform (Hardware/OS): Pi 3 Model B

Node version: 9.11.2

MagicMirror version: 2.4.1

Module version: 1.0.2

Description of the issue:
Hey, I would like some help installing the module.  I've tried numerous times, always kept getting one error or another. Decided to scrap the OS and start fresh in case I had changed some file I should not have while pursuing the error codes and their fixes. 

On this last install I got an error at the end of the install:
[image]

Continued with the instructions putting it in the config file:
```
                {
                    module: 'MMM-voice',
                    position: 'bottom_bar',
                    config: {
                                microphone: 1,
                                keyword: 'MAGIC MIRROR',
                                timeout: 15,
                            }
                },
```
AND now my MM screen is blank (just black).

Checked the pm2 log and got back:

pi@raspberrypi:~/MagicMirror $ pm2 logs MagicMirror
[TAILING] Tailing last 15 lines for [MagicMirror] process (change the value with --lines option)
/home/pi/.pm2/logs/MagicMirror-error.log last 15 lines:
0|MagicMir |     at Object.<anonymous> (/home/pi/MagicMirror/modules/MMM-voice/node_helper.js:369:3)
0|MagicMir |     at Module._compile (module.js:642:30)
0|MagicMir |     at Object.Module._extensions..js (module.js:653:10)
0|MagicMir |     at Module.load (module.js:561:32)
0|MagicMir | { Error: Cannot find module 'pocketsphinx-continuous'
0|MagicMir |     at Module._resolveFilename (module.js:543:15)
0|MagicMir |     at Function.Module._resolveFilename (/home/pi/MagicMirror/node_modules/electron/dist/resources/electron.asar/common/reset-search-paths.js:35:12)
0|MagicMir |     at Function.Module._load (module.js:473:25)
0|MagicMir |     at Module.require (module.js:586:17)
0|MagicMir |     at require (internal/module.js:11:18)
0|MagicMir |     at Object.<anonymous> (/home/pi/MagicMirror/modules/MMM-voice/node_helper.js:14:13)
0|MagicMir |     at Object.<anonymous> (/home/pi/MagicMirror/modules/MMM-voice/node_helper.js:369:3)
0|MagicMir |     at Module._compile (module.js:642:30)
0|MagicMir |     at Object.Module._extensions..js (module.js:653:10)
0|MagicMir |     at Module.load (module.js:561:32) code: 'MODULE_NOT_FOUND' }

/home/pi/.pm2/logs/MagicMirror-out.log last 15 lines:
0|MagicMir | Module helper loaded: MMM-Remote-Control
0|MagicMir | No helper found for module: clock.
0|MagicMir | Initializing new module helper ...
0|MagicMir | Module helper loaded: calendar
0|MagicMir | Initializing new module helper ...
0|MagicMir | Module helper loaded: MMM-quote-of-the-day
0|MagicMir | No helper found for module: currentweather.
0|MagicMir | No helper found for module: weatherforecast.
0|MagicMir | No helper found for module: worldclock.
0|MagicMir | Initializing new module helper ...
0|MagicMir | Module helper loaded: MMM-MTA-NextBus
0|MagicMir | Whoops! There was an uncaught exception...
0|MagicMir | MagicMirror will not quit, but it might be a good idea to check why this happened. Maybe no internet connection?
0|MagicMir | If you think this really is an issue, please open an issue on GitHub: https://github.com/MichMich/MagicMirror/issues
0|MagicMir | Launching application.

Commenting it out of the config file got my page up and running again w/out the voice module. Any clue as to what to do?

fewieden, 2018-07-26:
From your logs it looks like you executed the installer as sudo. `/root/MagicMirror/...`

Thats why he can't find the directory because it doesn't exist. Your stuff is located at `/home/pi/MagicMirror/...`

Don't run the installer as sudo

314noob, 2018-07-26:
Ok, I've been following the tips already given in [image] and arrived at this error in the debug log:
"pocketsphinx_continuous: error while loading shared libraries: libpocketsphinx.so.3: cannot open shared object file: No such file or directory"

Icon on screen says error undefined. 
I am once again lost.

fewieden, 2018-07-26:
try this one https://forum.magicmirror.builders/topic/6939/mmm-voice-setup-but-magicmirror-wont-display/28

314noob, 2018-07-26:
Thank you so much, that did it. 

It is up and running, although still mocking me, it will activate, write out the commands I say, but does not execute any of them.

314noob, 2018-07-27:
The debug shows it understands what I am saying, yet it does nothing.
[image]

I did get the icon to blink once, once, but it did not respond to the command that time either.

fewieden, 2018-07-27:
it will not run any command as long as you didn't select a mode https://github.com/fewieden/MMM-voice#select-mode

314noob, 2018-07-27:
I'm sorry to be a pest, but could you further explain that to me? 
I've tried saying "hide", "show", "wake", "go", "open" and "close" and have had no luck.

fewieden, 2018-07-28:
This module has the concepts of the different components:

- wake word
- mode
- commands

The wake word needs to be said every time the listener is timed out.

The mode needs to be the first word (except the first word is the wake word then it needs to be second), but only needs to be said once. The current mode gets displayed next to the microphone icon. When you want to change the mode to a different mode (https://github.com/fewieden/MMM-voice/wiki/Supported-Modules), it has to be the first word again. Commands will only be detected for the module of the current mode, to avoid conflicts that two modules have the same commands.

Once you are in a mode the mirror is listening to your commands that get send to the related module, which is parsing the command and acts on it.

I hope it is now more clear for you.

As long as you are not able to enter the voice mode (text voice next to the mic icon), the mirror never will act on your commands.

314noob, 2018-07-31:
Got it.  Thank you so much for your help.  
Sorry for the delayed response, life got very hectic the past few days.

closed 2018-07-31 by 314noob