fewieden/MMM-voice #26: ERROR: "dict.c"

TheWobbyDobbyDobDob, 2018-08-21:
Platform (Hardware/OS): Raspberry Pi 3

Node version: v9.11.2

MagicMirror version: newest 

Module version: newest

Description of the issue: Hey guys, after installing the module I had an Error 127 with the library problem that was already discussed. After spending the last two days ( I´m an absolute beginner) I "fixed" it. Now It says "1" in the error.log and the following in the debug.log:

[image]

hope you can help me.

TheWobbyDobbyDobDob, 2018-08-21:
and I use the latest Raspbian Stretch OS

fewieden, 2018-08-22:
@TheWobbyDobbyDobDob can you please tell me what files you have in the MMM-voice directory?

TheWobbyDobbyDobDob, 2018-08-23:
the following:

Bytes.js
CHANGELOG.md
debug.log
DEVELOPER.md
error.log
jsdoc.json
LICENSE
MMM-voice.css
MMM-voice.js
node_helper.js
package.json
README.md
words.json

fewieden, 2018-08-23:
Ok, for some reason it wasn't creating `MMM-voice.dic` and `MMM-voice.lm` files. Can you try to delete `words.json`, the `error.log` and the `debug.log` and then restart the mirror?

TheWobbyDobbyDobDob, 2018-08-23:
thanks for the tip, now it works 👍. But it doen´t seem to understand what I´m saying  :/

fewieden, 2018-08-25:
The mirror can only understand a certain amount of words, listed in the words.json. What are you trying to do, what are you saying and what is the mirror displaying that he recognizes?

TheWobbyDobbyDobDob, 2018-08-30:
Well after a few tries the mirror recognices the key word "MagicMirror" but afterwards I tried the IP application you developed and iht didn´t get any of the commands :/ could it simply be the microphone? it was quite cheap

fewieden, 2018-08-30:
Some people reported microphones, other had issues with their accents. One guy told me that he is even using a text2speech app on his phone because the mirror cant understand him.