fewieden/MMM-voice #28: Voice recognition

adbeem, 2018-09-19:
Hi first of all thank you for your awesome project!
Since english is not my first language i'm having trouble with my commands not being recognized properly. I just want to use simple functions such as "Go to sleep" or "wake up" but are there any ways for the commands to be adapted to my voice?

fewieden, 2018-09-20:
Pocketsphinx can be used in multiple languages, but isn't supported by this module.

You can try to have a look into the `MMM-voice.dic` file, there it shows how you need to pronounce it, so the mirror can understand it. This also helped me to being recognized by the mirror.

Another possibility is to set `debug: true` in the config, then it will be displayed what the mirror understands.

adbeem, 2018-09-24:
hi fewieden, thanks for your response. i got error when starting. i tried the two commands:
LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/usr/local/lib
sudo ldconfig
but nothing showed up. i check the file directory and i cant seem to find any ldconfig file (also in hidden mode). It starts from initializing then ready choose a mode and immediately goes to ERROR, i checked the error.log it shows 1 1 1 1 1 1 ...

fewieden, 2018-09-25:
Can you enable `debug: true` in the config, this will generate a debug log with more informations.

adbeem, 2018-09-26:
i just did a fresh reinstall of MMM-voice and got it working now. but yeah im Asian and my accent is very bad for this module. thank you anyway fewieden!