fewieden/MMM-voice #30: Cannot get mirror to recognize commands

shayfer, 2018-10-13:
Platform (Hardware/OS): Raspberry PI 3 

Node version: Newest build 10/13/2018

MagicMirror version: Newest build 10/13/2018

Module version: Newest build 10/13/2018

Description of the issue: Cannot get module working. Using microphone from Amazon that PI recognizes at ID 1. https://www.amazon.com/gp/product/B00UZY2YQE/ref=oh_aui_detailpage_o08_s00?ie=UTF8&psc=1

fewieden, 2018-10-14:
Can you share the content of your error log?

shayfer, 2018-10-14:
fewieden,

thanks for the reply!  Are the logs located at .npm?

fewieden, 2018-10-15:
the module has a dedicated error log in the modules directory

shayfer, 2018-10-15:
fewieden,

Thanks for the reply! I went through all the files in the module directory and I cannot find the log file.

[image]

fewieden, 2018-10-17:
Ok the error log is not there, that means the problem you have occurs even before the module is started. Can you check if you get errors in the terminal in which you run MM²?

shayfer, 2018-10-18:
fewieden,

Here is my mm-out.log, minus account info.

Starting MagicMirror: v2.5.0
Loading config ...
Loading module helpers ...
No helper found for module: alert.
Initializing new module helper ...
Module helper loaded: updatenotification
No helper found for module: clock.
Initializing new module helper ...
Module helper loaded: calendar
No helper found for module: currentweather.
No helper found for module: weatherforecast.
Initializing new module helper ...
Module helper loaded: MMM-RTSPStream
Initializing new module helper ...
Module helper loaded: newsfeed
Initializing new module helper ...
Module helper loaded: MMM-Wunderlist
No helper found for module: MMM-MotionDetector.
No helper found for module: MMM-voice.
All module helpers loaded.
Starting server on port 8080 ... 
Server started ...
Connecting socket for: updatenotification
Connecting socket for: calendar
Starting node helper for: calendar
Connecting socket for: MMM-RTSPStream
Connecting socket for: newsfeed
Starting module: newsfeed
Connecting socket for: MMM-Wunderlist
Sockets connected & modules started ...
Launching application.
Create new calendar fetcher for url: https://calendar.google.com/calendar/ical/BLANK - Interval: 300000
Create new news fetcher for url: http://feeds.bbci.co.uk/news/video_and_audio/news_front_page/rss.xml?edition=uk - Interval: 300000
MMM-Wunderlist: Create new todo fetcher for list: BLANK - Account: BLANK - Interval: 60000
MMM-Wunderlist: Create new todo fetcher for list: BLANK - Account: BLANK - Interval: 60000
MMM-Wunderlist: Create new todo fetcher for list: BLANK - Account: BLANK - Interval: 60000
Starting stream stream1 with args: [
    "--avdict",
    "rtsp_transport:tcp",
    "--win",
    "682, 1521, 1034, 1763",
    "--live",
    "--video_queue",
    "4",
    "--fps",
    "30",
    "rtsp://BLANK@192.168.1.78:554/videoMain"
]
PM2: omx_stream1 already running. Stopping old instance...
Starting PM2 for omx_stream1
PM2 started for omx_stream1

fewieden, 2018-10-18:
This sounds wrong to me `No helper found for module: MMM-voice.`, As you can see on your screenshot there is a node helper. Do you have something special in the config or put the module in a wrong place?

fewieden, 2018-10-21:
I think your problem is, that you downloaded the module instead of cloning it. Sorry that I didn't catch that on the first screenshot you send.

Loading modules depends on the name and your module directory name is wrong. You need to remove `-master`, then it should find the node helper.

Config module name has to match directory name.
Directory name has to match the file name inside.
The file name has to match the string inside the file `Module.register('MMM-voice', {`

By downloading a project on github, the branch name gets appended. If you clone it via `git clone` this doesn't happen.

shayfer, 2018-10-22:
fewieden,

Thanks and great catch! I re-imaged my PI and did a git clone https://github.com/fewieden/MMM-voice.git and proceeded with bash dependencies.sh. I was successfully able to get the microphone icon but it said error.  I checked the forums and saw this post which suggested running this command. LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/usr/local/lib
sudo ldconfig

This fixed the issue.  (On a side note after running the dependencies I did run a "npm install")

Everything is working great now! I just need to figure out how to Wake Up the mirror after I tell it to go to sleep. I wait a while and do the same procedure, Magic Mirror, Voice, Wake Up. but nothing happens.

Thanks for all your help!!

shayfer, 2018-10-22:
Resolved.  
sudo nano /boot/config.txt 
#Always force HDMI output and enable HDMI sound
hdmi_force_hotplug=1

closed 2018-10-22 by shayfer