fewieden/MMM-voice #31: Installing pocketsphinx | Failed

anmol12singh, 2018-10-23:
Platform (Hardware/OS): Raspberry Pi 3 Raspbian Stretch

Node version:

MagicMirror version: The latest one

Module version:

Description of the issue:
Attempting to install MMM-voice until an error occurs in step 3 saying "Installing pocketsphinx | Failed"

fewieden, 2018-10-23:
Can you only see this line or are there more detailed errors visible?

anmol12singh, 2018-10-24:
> Can you only see this line or are there more detailed errors visible?

Steps 1 And 2 Successfully install Until Step 3 is reached.
[STEP 3/5] Installing pocketsphinx
Cloning into 'pocketsphinx'...
remote: Enumerating objects: 12294, done.
error: RPC failed; curl 56 GnuTLS recv error (-9): A TLS packet with unexpected length was received.
fatal: The remote end hung up unexpectedly
fatal: early EOF
fatal: index-pack failed
[STEP 3/5] Installing pocketsphinx | Failed

fewieden, 2018-10-24:
It looks like `git clone` of pocket sphinx is failing, you might want to check out those suggestions https://stackoverflow.com/questions/38378914/git-error-rpc-failed-curl-56-gnutls

fewieden, 2018-11-18:
Closed due to inactivity.

closed 2018-11-18 by fewieden