fewieden/MMM-voice #39: Does Voice support a state of Pause?

TheStigh, 2019-02-01:
Platform (Hardware/OS): amd64/Ubuntu 16.04
Node version: 8.15
MagicMirror version: 2.6
Module version: Latest as of Feb 01 2019

Description of the issue:

Does MMM-Voice support receiving notification of Pause and Resume?
Reason for asking is making MMM-Voice send a notification to MMM-AssistantMk2 where this notification get a reply with Pause (configurable reply), whereas MMM-Voice need to stop listening on the microphone. 

(A workaround could be where a specific command just timeout the listening time of microphone).

AMK2 support replying with Pause and Resume (configurable replies).

fewieden, 2019-02-27:
@TheStigh this is currently not implemented. What is the reason behind it, that you're using multiple voice assistants?

TheStigh, 2019-03-03:
> @TheStigh this is currently not implemented. What is the reason behind it, that you're using multiple voice assistants?
Hi, this is resolved already :) Having the option to use offline or online assistants.

closed 2019-03-03 by TheStigh