fewieden/MMM-voice #40: Error 1

nabbi95, 2019-02-10:
Platform (Hardware/OS):Raspbian

Node version:

MagicMirror version:latest

Module version:latest 

Description of the issue: i hv tired so many things which you have told to troubleshoot those problems but it didnt work for me 
it shows error 1 in error log
and shows error in magic mirror too

nabbi95, 2019-02-10:
It shows "Ready to choose mode" for a second then goes to ERROR

nabbi95, 2019-02-10:
This is my debug.log

INFO: pocketsphinx.c(153): Parsed model-specific feature parameters from /usr/local/share/pocketsphinx/model/en-us/en-us/feat.params
Current configuration:
[NAME]			[DEFLT]		[VALUE]
-agc			none		none
-agcthresh		2.0		2.000000e+00
-allphone				
-allphone_ci		yes		yes
-alpha			0.97		9.700000e-01
-ascale			20.0		2.000000e+01
-aw			1		1
-backtrace		no		no
-beam			1e-48		1.000000e-48
-bestpath		yes		yes
-bestpathlw		9.5		9.500000e+00
-ceplen			13		13
-cmn			live		batch
-cmninit		40,3,-1		41.00,-5.29,-0.12,5.09,2.48,-4.07,-1.37,-1.78,-5.08,-2.05,-6.45,-1.42,1.17
-compallsen		no		no
-dict					modules/MMM-voice/MMM-voice.dic
-dictcase		no		no
-dither			no		no
-doublebw		no		no
-ds			1		1
-fdict					
-feat			1s_c_d_dd	1s_c_d_dd
-featparams				
-fillprob		1e-8		1.000000e-08
-frate			100		100
-fsg					
-fsgusealtpron		yes		yes
-fsgusefiller		yes		yes
-fwdflat		yes		yes
-fwdflatbeam		1e-64		1.000000e-64
-fwdflatefwid		4		4
-fwdflatlw		8.5		8.500000e+00
-fwdflatsfwin		25		25
-fwdflatwbeam		7e-29		7.000000e-29
-fwdtree		yes		yes
-hmm					/usr/local/share/pocketsphinx/model/en-us/en-us
-input_endian		little		little
-jsgf					
-keyphrase				
-kws					
-kws_delay		10		10
-kws_plp		1e-1		1.000000e-01
-kws_threshold		1e-30		1.000000e-30
-latsize		5000		5000
-lda					
-ldadim			0		0
-lifter			0		22
-lm					modules/MMM-voice/MMM-voice.lm
-lmctl					
-lmname					
-logbase		1.0001		1.000100e+00
-logfn					
-logspec		no		no
-lowerf			133.33334	1.300000e+02
-lpbeam			1e-40		1.000000e-40
-lponlybeam		7e-29		7.000000e-29
-lw			6.5		6.500000e+00
-maxhmmpf		30000		30000
-maxwpf			-1		-1
-mdef					
-mean					
-mfclogdir				
-min_endfr		0		0
-mixw					
-mixwfloor		0.0000001	1.000000e-07
[... 79 lines cut ...]
INFO: dict.c(213): Dictionary size 34, allocated 0 KiB for strings, 0 KiB for phones
INFO: dict.c(361): 5 words read
INFO: dict2pid.c(396): Building PID tables for dictionary
INFO: dict2pid.c(406): Allocating 42^3 * 2 bytes (144 KiB) for word-initial triphones
INFO: dict2pid.c(132): Allocated 21336 bytes (20 KiB) for word-final triphones
INFO: dict2pid.c(196): Allocated 21336 bytes (20 KiB) for single-phone word triphones
INFO: ngram_model_trie.c(354): Trying to read LM in trie binary format
INFO: ngram_model_trie.c(365): Header doesn't match
INFO: ngram_model_trie.c(177): Trying to read LM in arpa format
INFO: ngram_model_trie.c(193): LM of order 3
INFO: ngram_model_trie.c(195): #1-grams: 25
INFO: ngram_model_trie.c(195): #2-grams: 46
INFO: ngram_model_trie.c(195): #3-grams: 23
INFO: lm_trie.c(474): Training quantizer
INFO: lm_trie.c(482): Building LM trie
INFO: ngram_search_fwdtree.c(74): Initializing search tree
INFO: ngram_search_fwdtree.c(101): 28 unique initial diphones
INFO: ngram_search_fwdtree.c(186): Creating search channels
INFO: ngram_search_fwdtree.c(323): Max nonroot chan increased to 157
INFO: ngram_search_fwdtree.c(333): Created 28 root, 29 non-root channels, 5 single-phone words
INFO: ngram_search_fwdflat.c(157): fwdflat: min_ef_width = 4, max_sf_win = 25
INFO: continuous.c(307): pocketsphinx_continuous COMPILED ON: Feb 10 2019, AT: 22:28:23

Error opening audio device plughw:1 for capture: Device or resource busy
FATAL: "continuous.c", line 245: Failed to open audio device

TheStigh, 2019-02-11:
@nabbi95 You are sure you have set the correct microphone?

nabbi95, 2019-02-11:
@TheStigh  yess my microphone is working great because i have already installed Gassistpi in my Raspberry pi 
but error shows in MMM-voice module

TheStigh, 2019-02-11:
@nabbi95 What I mean, did you create an entry for microphone in config.js for MMM-voice? From the errorlog it expect the mic to be 1, you sure that is the correct setting?
Please paste a copy of your config entry for MMM-voice.

nabbi95, 2019-02-11:
Microphone has been set to 1 and yes Gassistpi is running also but it doesn't occupy microphone all the time it just occupy when we call that with a hotword otherwise it is free and my mmm-voice config file is

module: 'MMM-voice',
    position: 'top_center',
    config: {
        microphone: 1,
        }
    },

TheStigh, 2019-02-11:
@nabbi95 Gassist occupy the microphone all the time (as it is listening for the hotword).
I have not tested GassistPI yet, but it does not have an entry within config.js?

To troubleshoot, try to change microphone on GassistPI for testing purposes, lets say to '0' and then run MM, I guess MMM-voice will start up fine.

nabbi95, 2019-02-11:
ok i will try using mm-voice after stopping Gassist 
then i will let you know the report
thanks.

nabbi95, 2019-02-11:
i did it by closing gassistpi
it runs for the first time but didnt recognise my commands
but after restarting magicmirror it again starting showing error
and error log showing 1
and after turning debug to true it shows same error log as i uploaded above.
please help me

nabbi95, 2019-02-11:
INFO: pocketsphinx.c(153): Parsed model-specific feature parameters from /usr/local/share/pocketsphinx/model/en-us/en-us/feat.params
Current configuration:
[NAME]			[DEFLT]		[VALUE]
-agc			none		none
-agcthresh		2.0		2.000000e+00
-allphone				
-allphone_ci		yes		yes
-alpha			0.97		9.700000e-01
-ascale			20.0		2.000000e+01
-aw			1		1
-backtrace		no		no
-beam			1e-48		1.000000e-48
-bestpath		yes		yes
-bestpathlw		9.5		9.500000e+00
-ceplen			13		13
-cmn			live		batch
-cmninit		40,3,-1		41.00,-5.29,-0.12,5.09,2.48,-4.07,-1.37,-1.78,-5.08,-2.05,-6.45,-1.42,1.17
-compallsen		no		no
-dict					modules/MMM-voice/MMM-voice.dic
-dictcase		no		no
-dither			no		no
-doublebw		no		no
-ds			1		1
-fdict					
-feat			1s_c_d_dd	1s_c_d_dd
-featparams				
-fillprob		1e-8		1.000000e-08
-frate			100		100
-fsg					
-fsgusealtpron		yes		yes
-fsgusefiller		yes		yes
-fwdflat		yes		yes
-fwdflatbeam		1e-64		1.000000e-64
-fwdflatefwid		4		4
-fwdflatlw		8.5		8.500000e+00
-fwdflatsfwin		25		25
-fwdflatwbeam		7e-29		7.000000e-29
-fwdtree		yes		yes
-hmm					/usr/local/share/pocketsphinx/model/en-us/en-us
-input_endian		little		little
-jsgf					
-keyphrase				
-kws					
-kws_delay		10		10
-kws_plp		1e-1		1.000000e-01
-kws_threshold		1e-30		1.000000e-30
-latsize		5000		5000
-lda					
-ldadim			0		0
-lifter			0		22
-lm					modules/MMM-voice/MMM-voice.lm
-lmctl					
-lmname					
-logbase		1.0001		1.000100e+00
-logfn					
-logspec		no		no
-lowerf			133.33334	1.300000e+02
-lpbeam			1e-40		1.000000e-40
-lponlybeam		7e-29		7.000000e-29
-lw			6.5		6.500000e+00
-maxhmmpf		30000		30000
-maxwpf			-1		-1
-mdef					
-mean					
-mfclogdir				
-min_endfr		0		0
-mixw					
-mixwfloor		0.0000001	1.000000e-07
-mllr					
-mmap			yes		yes
[... 77 lines cut ...]
INFO: dict.c(213): Dictionary size 34, allocated 0 KiB for strings, 0 KiB for phones
INFO: dict.c(361): 5 words read
INFO: dict2pid.c(396): Building PID tables for dictionary
INFO: dict2pid.c(406): Allocating 42^3 * 2 bytes (144 KiB) for word-initial triphones
INFO: dict2pid.c(132): Allocated 21336 bytes (20 KiB) for word-final triphones
INFO: dict2pid.c(196): Allocated 21336 bytes (20 KiB) for single-phone word triphones
INFO: ngram_model_trie.c(354): Trying to read LM in trie binary format
INFO: ngram_model_trie.c(365): Header doesn't match
INFO: ngram_model_trie.c(177): Trying to read LM in arpa format
INFO: ngram_model_trie.c(193): LM of order 3
INFO: ngram_model_trie.c(195): #1-grams: 25
INFO: ngram_model_trie.c(195): #2-grams: 46
INFO: ngram_model_trie.c(195): #3-grams: 23
INFO: lm_trie.c(474): Training quantizer
INFO: lm_trie.c(482): Building LM trie
INFO: ngram_search_fwdtree.c(74): Initializing search tree
INFO: ngram_search_fwdtree.c(101): 28 unique initial diphones
INFO: ngram_search_fwdtree.c(186): Creating search channels
INFO: ngram_search_fwdtree.c(323): Max nonroot chan increased to 157
INFO: ngram_search_fwdtree.c(333): Created 28 root, 29 non-root channels, 5 single-phone words
INFO: ngram_search_fwdflat.c(157): fwdflat: min_ef_width = 4, max_sf_win = 25
INFO: continuous.c(307): pocketsphinx_continuous COMPILED ON: Feb 10 2019, AT: 22:28:23

Error opening audio device plughw:1 for capture: Device or resource busy
FATAL: "continuous.c", line 245: Failed to open audio device

fewieden, 2019-02-27:
@TheStigh thanks for helping here.

@nabbi95 are you sure that you closed all instances of magicmirror and gassistpi? Delete the error and debug log. Then start the magicmirror. What are the logs containing now?

closed 2021-01-15 by fewieden