fewieden/MMM-voice #41: ERROR 127

kshatriypiyush, 2019-02-26:
Platform (Hardware/OS):raspberry pi

Node version:

MagicMirror version:

Module version:

Description of the issue:
The microphone initializes then error is shown in error log it says 127

fewieden, 2019-02-27:
@kshatriypiyush add the flag `debug: true` in the module directory. Then start your mirror. There will be a debug log file generated in the module directory. What is the content?

kshatriypiyush, 2019-02-27:
i entered debug:true in debug diractory but on starting in mirror undefined error is written beside the mic symbol and in error log it shows 127 again..
here is my debug log file
[debug.log](https://github.com/fewieden/MMM-voice/files/2910886/debug.log)

fewieden, 2019-02-28:
@kshatriypiyush `Failed to open dictionary file `MMM-voice.dic` for reading: No such file or directory`

Go to the module directory and delete the following 3 files if existing:
- MMM-voice.dic
- MMM-voice.lm
- words.json

kshatriypiyush, 2019-04-11:
I did as u said removed all the three files.
MMM-voice.dic
MMM-voice.lm
words.json
 but still the error undefined is shown beside the microphone symbol and in debug file it says the same thing i.e. Failed to open dictionary file MMM-voice.dic for reading: No such file or directory

caphill7, 2019-06-04:
I had this issue, and solved it by adding the following line to ~/.profile
export LD_LIBRARY_PATH=/usr/local/lib

then reload the profile with
source ~/.profile

closed 2019-09-11 by fewieden