fewieden/MMM-voice #42: Change language

hugo82000, 2019-03-02:
Platform (Hardware/OS): Raspberry PI 3

Node version: LAST

MagicMirror version: LAST

Module version: LAST

Description of the issue: Hello, I'm trying to change the language so I modified the file .dic but you said that you use an online version of pocketsphinx so I think that modify files .dic, .lm, words.json etc..
I did the debug : true but I don't see the words I had added even if I see the in the file .dic
Can you help me please?

hugo82000, 2019-03-02:
I see that in the node_helper you use a function which generate dictionairy and language model but can I just use my files .dic and lm in french for exemple (If I change the code in MMM-voice.js for understanding my words and commands)?

Code in node helper
/**
     * @function generateDicLM
     * @description Generates new Dictionairy and Language Model.
     */
    generateDicLM() {
        console.log(`${this.name}: Generating dictionairy and language model.`);

        fs.writeFile('modules/MMM-voice/words.json', JSON.stringify({ words: this.words }), (err) => {
            if (err) {
                console.log(`${this.name}: Couldn't save words.json!`);
            } else {
                console.log(`${this.name}: Saved words.json successfully.`);
            }
        });

        lmtool(this.words, (err, filename) => {
            if (err) {
                this.sendSocketNotification('ERROR', 'Couldn\'t create necessary files!');
            } else {
                fs.renameSync(`${filename}.dic`, 'modules/MMM-voice/MMM-voice.dic');
                fs.renameSync(`${filename}.lm`, 'modules/MMM-voice/MMM-voice.lm');

                this.startPocketsphinx();

                fs.unlink(`${filename}.log_pronounce`, this.noOp);
                fs.unlink(`${filename}.sent`, this.noOp);
                fs.unlink(`${filename}.vocab`, this.noOp);
                fs.unlink(`TAR${filename}.tgz`, this.noOp);
            }

fewieden, 2019-03-04:
It only generates new .dic and .lm files if the words.json is not matching the required words. So if you start the mirror it will generate the words.json and .dic and .lm file. As long as you don't change your config.js it will not generate those files again. Then you can replace the lm and dic file. DO NOT CHANGE WORDS.JSON. Otherwise the .dic and .lm file are regenerated.

I'm not sure though if this will help you to use french or if you need to have some other parameters during the installation too. Also the modules are checking for english words.

hugo82000, 2019-03-04:
I will try whitout modify words.Json and i Will tell you If i success to manage it correctly

hugo82000, 2019-03-05:
When I change the file.dic just with the words I want, it replace it with adding some words like "the", "on","wall" is it normal ? I didn't change words.json like you said to me

fewieden, 2019-03-05:
@hugo82000 I assume that you need to change the dic and lm file

hugo82000, 2019-03-11:
I changed it but when I launch MagicMirror it change automaticly with some other words...

closed 2021-01-15 by fewieden