fewieden/MMM-voice #46: Getting ERROR next to mic when running MagicMirror

jmart1009, 2019-09-30:
Platform (Hardware/OS):Stretch

Node version:

MagicMirror version: 2.8

Module version:newest

Description of the issue:No matter what i do the software doesn't seem to be able to pic up my mic or something, have troubleshooted a lot on the MM forum, but just can't seem to get it to work. My mic works and have tested it as plughw:1,0, and i have changed the mic config in the files to be that way, but it still won't work. Don't know if there is another issue or not but would greatly appreciate help. Thanks in advance.
[debug.log](https://github.com/fewieden/MMM-voice/files/3668510/debug.log)
[error.log](https://github.com/fewieden/MMM-voice/files/3668513/error.log)

fewieden, 2019-10-05:
The error that is thrown is (bottom of the debug log):

```
Error opening audio device plughw:1,0 for capture: Invalid argument
FATAL: "continuous.c", line 245: Failed to open audio device
```

Failed to open audio device often is raised because something else on the pi is using the mic already. Make sure nothing else is using it and retry.

mujeebishaque, 2019-11-04:
I have checked no other process is using the mic, but it still doesn't work. I've tried changing the microphone attribute to 0 and 1, doesn't work either.

bjorndegroot, 2020-04-25:
Can anybody help with this issue? I've the same here.

fewieden, 2020-04-28:
@bjorndegroot what's inside your error and debug log?

closed 2021-01-15 by fewieden