fewieden/MMM-voice #51: Undefined Microphone Error

spencerandhill, 2021-02-12:
[image]
Platform (Hardware/OS):
Raspberry Pi 3 Model B Rev 1.2
Raspbian Stretch
Linux raspberrypi 5.4.51-v7+ #1333 SMP Mon Aug 10 16:45:19 BST 2020 armv7l GNU/Linux

Node version:
v10.23.3

MagicMirror version:
MagicMirror: v2.14.0

Module version:
1.0.2

Description of the issue:
After Installing a fresh Raspberry Pi with a fresh Magic Mirror and all the dependencies for this module, I cannot get it to work.
I configured only these modules:
alert, clock, MMM-voice
After starting the Magic Mirror and opening the URL, it only shows "ERROR" in the lower section. Find my config attached
[config.js.txt](https://github.com/fewieden/MMM-voice/files/5972750/config.js.txt)
(I renamed the config to config.js.txt as I couldn't upload it as config.js)

**My Audio Hardware**
I'm using the "ReSpeaker 2-Mics Pi HAT" Module for Microphone. I installed everything from https://wiki.seeedstudio.com/ReSpeaker_2_Mics_Pi_HAT/ and I can successfully hear myself, when I try "arecord -f cd -Dhw:2 | aplay -Dhw:2"
The config also contains "microphone: 2".

I wasn't able to debug some information (because the Chrome Developer Tools didn't gave me any hint and I have no idea how to enable debugging for "npm run start" or "npm run server"

Any idea, what's going wrong here?

fewieden, 2021-02-13:
@spencerandhill thank you for the detailed description. On a first glance the config seems right. To get more details you can run the module as:

```js
{
  module: 'MMM-voice',
  position: 'bottom_bar',
  config: {
    debug: true,
    microphone: 2,
    keyword: 'MAGIC MIRROR',
    timeout: 15
  }
}
```

This will generate a debug and error log in the module directory with more informations.

spencerandhill, 2021-02-15:
@fewieden thanks for your fast response! I got the log and also some hints.

> Error opening audio device plughw:2 for capture: Device or resource busy
> FATAL: "continuous.c", line 245: Failed to open audio device

And later:

> INFO: continuous.c(275): Ready....
> INFO: continuous.c(261): Listening...

Unfortunately I have no idea, **why** it can't access the audio device.

The full debug.log is attached here
[debug.log](https://github.com/fewieden/MMM-voice/files/5983407/debug.log)
 and the error.log here
[error.log](https://github.com/fewieden/MMM-voice/files/5983408/error.log)
(Error Log only shows a couple of lines with the number "127", nothing else)

fewieden, 2021-02-16:
@spencerandhill resource busy usually was reported when a user had something else running that accessed the microphone.

You could try to run pocketsphinx standalone first without the mirror. 

`pocketsphinx_continuous -adcdev plughw:2 -inmic yes -lm PATH_TO_LM_FILE -dict PATH_TO_DIC_FILE`

This is how the module is running it.

soubhik-khan, 2021-05-17:
Hi, I have a similar issue, it says 
`ERROR
undefined`

Platform (Hardware/OS): RPI 4B 2GB

My config
`                {
                            module: 'MMM-voice',
                            position: 'bottom_bar',
                            config: {
                                debug: true,
                                microphone: 2,
                                keyword: 'MAGIC MIRROR',
                                timeout: 15
                                }
                }`

error.log shows
`127`

debug.log shows
`pocketsphinx_continuous: error while loading shared libraries: libpocketsphinx.so.3: cannot open shared object file: No such file or directory`
I checked the file is there in /usr/local/lib
I also tried writing this in .bashrc but no luck
`export LD_LIBRARY_PATH=/usr/local/lib
export PKG_CONFIG_PATH=/usr/local/lib/pkgconfig`

Appreciate any help with this!

fewieden, 2021-05-19:
@soubhik-khan try this forum post https://forum.magicmirror.builders/post/35364

maku1958, 2021-12-29:
Hi @fewieden , I have the same problem.
I have run the debug log and in the error.log is one Error several times:
  GNU nano 5.4                        error.log                                 
Error: spawn pocketsphinx_continuous ENOENT
-2
Error: spawn pocketsphinx_continuous ENOENT
-2
Error: spawn pocketsphinx_continuous ENOENT
-2
Error: spawn pocketsphinx_continuous ENOENT
-2
Error: spawn pocketsphinx_continuous ENOENT
-2
Error: spawn pocketsphinx_continuous ENOENT
-2
Error: spawn pocketsphinx_continuous ENOENT
-2
Error: spawn pocketsphinx_continuous ENOENT
-2
Error: spawn pocketsphinx_continuous ENOENT
-2
Error: spawn pocketsphinx_continuous ENOENT
-2

This is in my folder:
pi@raspberrypi:~/MagicMirror/modules/MMM-voice/pocketsphinx $ ls
AUTHORS       include    Makefile.am         pocketsphinx.sln  swig
autogen.sh    indent.sh  model               README            test
configure.ac  LICENSE    NEWS                README.md         win32
doc           m4         pocketsphinx.pc.in  src

The command: pocketsphinx_continous is not found

Manfred

fewieden, 2022-01-21:
@maku1958 this sounds like something went wrong with the installation and he can't find the binary file.

I suggest deleting the module directory and starting from scratch.

With your debug.log file, this could be an older message. If you are sure the file exists it is usually a good idea to delete the log files again before the restart so you only see the still relevant entries.

maku1958, 2022-01-21:
Thanks. I will start with a complete new installation. I need first a new SD card. --Diese Nachricht wurde von meinem Android Mobiltelefon mit GMX Mail gesendet.